makerbase-mks/SGEN_L #1: ENDER 3 stock LCD NOT working, NO power LED

jarjartee, 2019-06-22:
Just received my Gen L board , flashed firmware in Arduino - all works apart from when connecting to power from 24V PSU - NO LED but fans are going. Tried 2 different tutorials to get LCD working BUT NOGO

pattonwebz, 2019-06-23:
I had to flip my display ribbon upside down on screen side (and cut the tab so it could go in the wrong way) to get it to work.

jarjartee, 2019-06-23:
Tried all that , problem is NO power LED when connecting to 24v which indicates a problem 
[image]
somewhere - USB connection is fine on its own but does NOT WORK whilst hooked up to power.

pattonwebz, 2019-06-24:
Sorry I misunderstood your issue originally. I don't know anything to help with your current problem sadly :(

thisiskeithb, 2019-06-26:
The [current pins file](https://github.com/makerbase-mks/SGEN_L/blob/05350a1864cafee9987f52e10155980adfbb508f/Firmware/Marlin-bugfix-2.0.x/Marlin/src/pins/pins_MKS_SGEN_L.h) in this repo is not setup for the Ender-3 display (`CR10_STOCKDISPLAY`), but look for board support in the main Marlin repo soon!

jarjartee, 2019-06-26:
Patton - YOUR're a STAR - swapping the end stops did the trick - there's life in the display now as well BUT not quite right configured - thanks Keith

closed 2019-06-26 by jarjartee

thisiskeithb, 2019-06-26:
Here's a link to the PR in Marlin's repo with Ender-3/`CR10_STOCKDISPLAY` (and several others) support: https://github.com/MarlinFirmware/Marlin/pull/14411

pattonwebz, 2019-06-26:
Happy to have helped and I am glad you shared that picture so we could figure out what was up :)

thisiskeithb, 2019-06-26:
@jarjartee: The SGEN-L is now supported upstream in Marlin, so give it a test.

pattonwebz, 2019-06-26:
@thisiskiethb with the new changes do you think that will mean all of us that flipped our cables will need to flip them back round too? I guess it will bit figured I'd ask if you know since I'm unlikely to do any flashing of an updated marlin anything this week.

thisiskeithb, 2019-06-26:
You'll have to continue to rotate your cables on MKS boards.

pattonwebz, 2019-06-26:
Good that I asked then before trying. Keeping cable flipped suits me fine. Thanks for the info :)