makerbase-mks/SGEN_L #3: big issue with the board

saliery999, 2019-08-30:
i use sgen_L with 2209 drivers (spi mode, smoothieware firmware) and mks tft 35, after start printing if you click stop while printing anything, board no response, also if print finished, board have no response and i cant start print anything before take off and on 3d printer. i analyse that the M84 command make board to no response. i thing there are bug in firmware.

saliery999, 2019-08-30:
also i triend any drivers, such as drv8825, anyway board no response

saliery999, 2019-08-31:
ok i think i found problem, in this board on extruder E0(EN) you use 2.12 pin same as smoothie ware use pin 2.12 for kill-pause-button, in config file (pause_button_enable (true)) i change to false and everything work fine. i can use any smoothieware firmware now too. maybe need change pin for this board for kill-pause-button in config file to any free 

https://drive.google.com/file/d/1NCUExftWjNEbVH38PPVC9itnEu5RRkMK/view
https://ibb.co/1RmDwz9

acethanhtung, 2019-09-06:
Hey guy. You can share me dataaheet mks sgen L? Thank you.

saliery999, 2019-09-06:
https://github.com/makerbase-mks/Datasheet/blob/master/English%20datasheet/MKS%20SGEN_L%20Datasheet.pdf

borland1, 2019-09-29:
Windows 10 Device Driver (Marlin/src/HAL/HAL_LPC1768/win_usb_driver) is not digitally signed.  Windows 10 Device Manager shows device as disk drive “Marlin SD card 01 USB Device” and USB Serial Device (COM4).  Manually installing driver says device driver currently installed is latest driver.  Forcing device driver to install gives Windows error stating "driver not digitally signed".

jmarton2, 2019-11-21:
Hey saliery999, may i ask you to share your config file? Thank you!

makerbase-mks, 2020-02-29:
Yes, you need to disable  #panel.back_button_pin function and my config.txt has disable it.
[image]

closed 2020-04-16 by makerbase-mks