makerbase-mks/SGEN_L #7: Marlin Firmware SD Card Issue

kusuprfn, 2019-11-06:
I have flashed the board with an unmodified version of the marlin firmware in this repository, and once flashed, the SD card slot seems to no longer function. LED lights 2-4 do not light up and LED 1 blinks constantly. Additionally the USB drive only seems to briefly recognize via USB connection when the board is booting, but then is no longer accessible nor can I flash the firmware again without using the reset button.

mrhempman69, 2019-11-07:
I have a couple of these boards. That is an indication that the board is working? What did you expect to happen?

Can you comment on my issue that I'm about to link to below?

makerbase-mks, 2019-11-08:
1. Marlin firmware only defines one LED on the motherboard for status display, this is normal.
2. Currently, the 4 LEDs reserved by MKS SGEN_L are only displayed when using Smoothieware firmware.

closed 2019-11-26 by makerbase-mks