makerbase-mks/SGEN_L #8: Print from SDcard on boot

mrhempman69, 2019-11-07:
Is it possible to print from the SD card on boot using these boards?

Normally, marlin firmware uses a file called `autostart.g` from the SD card and prints from that on boot. 

But when I include `autostart.g0` in the root of my board, the SGEN_L doesn't start printing from it.

Is the SD card only used for firmware flashing? 

Is there any way to print from this board without using a USB connection? 

Possibly related: https://github.com/makerbase-mks/SGEN_L/issues/7

makerbase-mks, 2019-11-08:
1. It is very certain to tell you that SD is supported for automatic printing;
2.gocde file name must be auto0.g instead of autostart.g
3. You re-edit the gcode file name to try, good luck!

biocrypto730, 2019-11-10:
Thanks. In my own custom firmware, I did not have `#define SDSUPPORT`. But this firmware does have sd support enabled by default. I thought it was a problem with the board.

mrhempman69, 2019-11-10:
Thanks. In my own custom firmware, I did not have #define SDSUPPORT. But this firmware does have sd support enabled by default. I thought it was a problem with the board.

closed 2019-11-10 by mrhempman69