makerbase-mks/SGEN_L #9: CoreXY setting does not work with SGEN_L + smoothieware

jmarton2, 2019-11-21:
I would like to use SGEN_L with 2209 drivers (normal mode, smoothieware firmware) for a corexy printer. I added the "arm_solution corexy" parameter to config.txt, but it has no effect, XY motors are working independently. :( 
I flashed the latest smoothieware firmware from here, it did not help.
Same config works well on SBASE v1.3 board.

makerbase-mks, 2019-11-25:
MKS SGEN_L and MKS SBASE, the two motherboard pins are not compatible, you need to modify the pin define in config.txt. You can refer to the MKS SGEN_L pin map: https://github.com/makerbase-mks/SGEN_L/tree/master/Hardware/MKS%20SGEN_L%20V1.0_001

jmarton2, 2019-11-25:
That is clear for me and I made the configuration accordingly. X, Y, Z and E0 steppers are working well independently on SGEN_L, but the "arm_solution corexy" parameter means the X and Y motors should work together. In case of corexy mechanism if just one motor works then the head moves diagonally. https://corexy.com/theory.html
This parameter is working on my SBASE v1.3 board in my corexy printer, I just wanted a quieter driver for it so I selected the SGEN_L board. Smoothieware firmware provided here is much smaller for SGEN_L board than for SBASE board (299k vs. 387k).
Could it be the reason for it? Is something related missing from it?

makerbase-mks, 2019-11-28:
[SGEN_L_LCD12864.zip](https://github.com/makerbase-mks/SGEN_L/files/3899885/SGEN_L_LCD12864.zip)
Can you test the firmware I provided? But  you may need to add Corexy definitions, and Corexy homing ...Becaus this XYZ machine configuration.

jmarton2, 2019-11-28:
That is the same firmware what is available here: https://github.com/makerbase-mks/SGEN_L/tree/master/Firmware/Smoothieware/MKS%20LCD12864. Firmware files are identical. This is what I have tried already. CoreXY definition and homing properly configured, as I wrote similar config is working with my SBASE v1.3 board.

makerbase-mks, 2020-02-29:
I think your corexy_homing have not be set in config.txt。like this picture
[image]

DanielSantosD, 2020-07-20:
Hello, i have the same problem using smoothiware the corexy arm solution doesant work can you help

mks-viva, 2020-10-20:
I will close this question temporarily.

closed 2020-10-20 by mks-viva