makerbase-mks/SGEN_L #10: Bl touch

JUJU3D, 2019-11-30:
Hello any one know how to use bot touch with mks sgenL what IS a good pin,thanks a lot

Sharkys80, 2019-12-05:
The pins for Bl Touch : 

[image]

crazy72, 2019-12-22:
Be careful in case you are upgrading from a stock MKS. Brown is GND and Red 5V. Sharkys80 are you sure you have to conenct the white and black to Z+ ? in the previous board everything is connected to X- , Y- and Z-

Sharkys80, 2019-12-23:
Correct its Zmin

[image]

CaptainRoot, 2020-01-17:
I upgraded my last weekend, i used z-pin without any problems

makerbase-mks, 2020-02-29:
We has about how to use 3D TOUCH video tutorial, you can refer to：https://www.youtube.com/watch?spm=a2g0o.detail.1000023.1.2b603d9b93YISB&v=4tmdXvN0K4U

closed 2020-03-03 by makerbase-mks