makerbase-mks/SGEN_L #11: No Pin Assignments, For Display MKS_12864OLED_SSD1306

borland1, 2019-12-08:
In Folder/File:   Marlin/src/pins/pins_MKS_SGEN_L.h

MKS_12864OLED_SSD1306 display is not supported because there are no pin assignments for this display to be enabled in Configuration.h file.

File MKS_SGEN_L.h should be modified to add something like this..

   #if EITHER(MKS_12864OLED, MKS_12864OLED_SSD1306)
        #define LCD_PINS_DC     P0.17   // Set as output on init
        #define LCD_PINS_RS      P1.00   // Pull low for 1s to init
        // DOGM SPI LCD Support
        #define DOGLCD_CS       P0.16
        #define DOGLCD_MOSI   P0.18
        #define DOGLCD_SCK     P0.15
        #define DOGLCD_A0       LCD_PINS_DC
        
        #define BEEPER_PIN      P1.31
        #define BTN_ENC         P1.30
    #endif

borland1, 2019-12-08:
I submitted an Issue Report (#16142) on the main Marlin Firmware depository.  They provided a working firmware.bin file based on my Configuration.h file.  They should be submitting PR fix soon.
Marlin has formally released firmware Version 2.0.x.  Recommend this MKS fork be updated with latest Marlin release.

makerbase-mks, 2019-12-12:
MKS OLED12864 display function has been added and has been tested.

closed 2019-12-12 by makerbase-mks