makerbase-mks/SGEN_L #15: When using MKS LV8729, board will not boot anymore

CaptainRoot, 2020-01-17:
Hello,
i bought some MKS LV8729 in order to use them as extruder driver. 

When i plug in the driver to my board an power up, it will not boot anymore. All diag led lights up but nothing happens. 

After removing the lv8729 and power cycling the board boots 

Any suggestions?

borland1, 2020-01-18:
Exactly how did you configure the driver jumpers?  MS1, MS2, and MS3 are mislabeled on the board as MS0, MS1, MS2.
Also, does the MKS LV8729 support 5V or 3.3V logic? If you look at the board schematic diagram, there is a jumper to supply the driver's VDD supply voltage with either 3.3V or 5V.

CaptainRoot, 2020-01-18:
Hi,
I used 
1/32 Microstep, High Low High, Jumper Free Jumper. Even with out any jumper the Board just light up all leds but will not boot. I use Marlin 2.0 Bugfix

borland1, 2020-01-18:
Looks like from photos I found for that driver, there are no silkscreen pin labels.  How did you decide how to orient the driver before inserting it on the board?

CaptainRoot, 2020-01-18:
i used a multimeter in order to find the VM ( V motor) pin. I use 12V for heater and stepper, and 5V logic voltage

borland1, 2020-01-18:
OK.  I see a diagram now that shows EN pin is located opposite end of the driver's POT for adjusting reference voltage Vref.  So, when driver is inserted in board, the POT should be on side closest to edge of SGEN board where reset button and ESP1/ESP2 are located.

CaptainRoot, 2020-01-18:
[image]

on the picture Vmotor is in the upper right corner. i guess i plugged the stepper the right way
VMotor on the chip is in the lower left corner pin 1 of the chip, 

On the mks sgen l the pot should be not facing to the  EXP1/EXP2 connector.

borland1, 2020-01-18:
Yes, but if you use your multimeter to trace continunity between the chip pin 1 and the header pin labeled VMOT on the SGEN board, you will see that you have the driver module inserted wrong.  Turn it around so the POT is towards EXP1/EXP2.

CaptainRoot, 2020-01-19:
[image]

This is the pinout i traced with the Multimeter. I have not idea why they place the chip on the underside of the pcb.

borland1, 2020-01-19:
OK, you must have it inserted correctly.  The driver designer must have decided the green and black color coding of the pin headers would eliminate the need for silkscreen pin labels.
Unless, there is a direct short of the driver module, there is no reason that the module would cause the SGEN board to boot.  Must be a bad batch of drivers.
LV8729 drivers are not highly rated, suggest you look for MKS TMC 2209 drivers which are finally starting to coming down in price.  Currently, MKS TMC 2209 can be had on AliExpress each for about $5.00USD when shipped to US.  There are other brands, but those will require soldering to adapt to this board for serial control, so best to use MKS brand TMC 2209.

CaptainRoot, 2020-01-19:
maybe the stepper driver is defect. I used TCM2209 before in speadcycle and/or stealthcop mode but when printing fast with retract it lost steps, Si i decided to change the driver to LV8729 bit wih no success. The A4988 worked perfect before but they are very loud and only have 1/16 microsteps. 

Now i use ST820 (STSPINS820) in 1/32 microstep mode , these are greate drivers, littlebit  complicat to set vref but very silent almost silent as the TMC

Thanks, for your help!

Sharkys80, 2020-01-22:
> maybe the stepper driver is defect. I used TCM2209 before in speadcycle and/or stealthcop mode but when printing fast with retract it lost steps, Si i decided to change the driver to LV8729 bit wih [...]

For TCM2209 stop the shifting you need the tmc debug menu +Hybrid Threshold (try 65 -75 for x,y) and the corect mA for each Motor (use the feeltest motor its self to hot  lower the mA) and dont forget cool down the driver withe a fan

CaptainRoot, 2020-01-22:
Hi Sharkys80,

on x, y i use TMC2208 ond on z one TMC2209. They work just perfect an yes they are cooled by a 5015 blower fan + fan tunnel. 

Only for the extruder i had no success with the TMCs, so i was looking for an other stepper alternative. The LV8729 are often recommended, so i gave them a try.

Now i use ST microsystems ST820 (STSPINS820).  Feels like no one uses these stepper drivers in the whole internet but they work great!

Sharkys80, 2020-01-22:
Hello CaptainRoot
I self use TMC2208 and TMC2130 on diffrent extruders all works fine.
The most difficulty is the titan direct the small motor max 680mA or it will  very hot, with Hybrid Threshold 25 and the motor dosent loses any stepps.And only for the extruder dont higher the MICROSTEPS over 32 more make no sense.You cannot raise the microns through it,but the Problems ;)

CaptainRoot, 2020-01-23:
Hi,
this is little of topic. But i am not the only one who has no success when using tmc drivers for the extruder stepper. i use a non geared  bowden extruder on my CR10 with the standard creality extruder motor. And i lost steps, so i decided to switch to another driver an now i am happy with it

mks-viva, 2020-02-23:
Hi, for this question. Initial confirmation of the e0_enable signal caused the motherboard to fail to start. At this stage, the E0 axis can use other types of drives, such as MKS TMC2209. You can use standard mode or Uart mode. At present, MKS TMC2209 supports higher current than 8729 ...

closed 2020-08-19 by makerbase-mks