makerbase-mks/SGEN_L #18: SGEN-L + MKS TMC2130 TMC connection error

flygja, 2020-02-29:
Hi I bought a SGEN-L and TMC2130 from MKS for X, Y, Z and E0. I followed the user guide to plug in 4 jumpers on the left side for each driver. I have compiled and uploaded Marlin 2.0-bugfix from MKS github. I can't get the motor drivers to work. The LCD screen shows TMC connection error and when I send M122 I get the error below.

Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH
Testing Z connection... Error: All HIGH
Testing E connection... Error: All HIGH

Temp reading, endstop switches, hotend/bed heater are working. Any idea what could be the issue?

mks-viva, 2020-03-01:
Hi, please provide your firmware and motherboard wiring diagram.

flygja, 2020-03-01:
I managed to get it working with Marlin 2.0.4.4. Have to enable "TMC_USE_SW_SPI" in configuration_adv.h. Please add that into the user manual as well. Thanks

mks-viva, 2020-03-01:
OK. It seems pretty good.

Crazor, 2020-12-04:
I've got the same issue. No matter if I'm enabling TMC_USE_SW_SPI or not, M122 says All HIGH. I've got TMC2130 modules from MKS. Do I need to do anything special except setting the four green jumpers under each driver and the 3.3V jumper?

mks-viva, 2020-12-04:
You need connect 12-24V power

Crazor, 2020-12-04:
Of course I have 12V hooked up, just didn't mention it.

mks-viva, 2020-12-04:
You need power on it of Dc12-24V

Crazor, 2020-12-04:
As I stated, I have 12V hooked up as well.

mks-viva, 2020-12-04:
Can you provide your wiring and your firmware to me, and i will test it, but it may be tomorrow, because for me, it is night.

Crazor, 2020-12-04:
On the bottom side of each driver board, there are two 0 ohm jumpers labeled CFG5 and CFG4. CFG5 (the one on top) is soldered on the left pad, CFG4 (the one near the edge of the board) is soldered on the right side. Is that correct for SPI mode?

mks-viva, 2020-12-04:
I want to know, where did you buy your TMC2130, because I know that MKS does not currently sell TMC2130 drivers.

Crazor, 2020-12-04:
I bought them about a year ago off of MKS's AliExpress store, IIRC. I didn't use them until now.

mks-viva, 2020-12-04:
Okay, then, I want to tell you that MKS TMC2130 is in SPI mode by default, you can use it directly by plugging it into the motherboard.

mks-viva, 2020-12-04:
Can you provide your firmware to me?

Crazor, 2020-12-04:
[The complete firmware tree I use](https://github.com/Crazor/Marlin/tree/2.0.x-MKS-SGEN_L)

The relevant config files:
[Configuration.h](https://github.com/Crazor/Marlin/blob/2.0.x-MKS-SGEN_L/Marlin/Configuration.h)
[Configuration_adv.h](https://github.com/Crazor/Marlin/blob/2.0.x-MKS-SGEN_L/Marlin/Configuration_adv.h)

mks-viva, 2020-12-04:
Ok! I will test it, but it may be tomorrow.

Crazor, 2020-12-04:
Nevermind, I got this working now. It was a combination of bad software settings and a bad driver module. Got 4 out of 5 working, must have fried the fifth one...

For anyone wondering: Put in four jumpers in the green position below every TMC2130 driver module, put the drive IC power jumper (below the Z driver module) on 3.3V.

As for software: 
Define X_DRIVER_TYPE  TMC2130 (and any other axis you want to use) in Configuration.h, define TMC_USE_SW_SPI in Configuration_adv.h (but none of the pins below that line, they are defined correctly with the board's pins), and define TMC_DEBUG if you want to look at the driver's registers with M122.

mks-viva, 2020-12-05:
It is will working, it is perfect. So, i will close this issues.

closed 2020-12-05 by mks-viva