makerbase-mks/SGEN_L #19: ISSUES WITH MKS SGEN_L + TFT24

jpoyatos1961, 2020-03-04:
Hi,

I'm new on here and I would like to know if somebody can help me, just had bought a MKS SGEN_L + TFT24, I don't know what is going on, but for some reason, the icons are missing on the TFT, or just part of it appears on the TFT.
My question is how to do the upgrade in the right way, I follow the instructions but still having the same problems.
This is what I'm doing if somebody can give me directions to figure out the problem, I'll really appreciate it.
1/ Download the firmware from this page (motherboard and TFT)
2/ Put the files for the TFT, config, mks_pics, mks_fonts and firmware, introduce the 2GB SD card on the reader beside TFT, press reboot make all the progress, and finally the same result, all icons with letters but white and not icons.
3/ Motherboard, download firmware from this web site, I'm using the firmware from the Smoothie folder, save that on the MSD card and put it on the MSD slot on the motherboard reboot the card LEDs blinks as the guide said, but nothing happens, also the CONFIG file never change from TXT to CURSOR extension, also I try to change the IP address or language or any other parameter I change on the config TXT file, nothing happens.

I'm very unhappy due to trying, this is my first steep and is a nightmare, seems to be easy on all I had read but practically never had any success.

Thanks for all the help you can give me.

Jose

fireflylr, 2020-03-05:
Tft24 related files download links for https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT2.4, you can download the files and then try to update try. The hope can help you.
                                                                       --mks

makerbase-mks, 2020-04-16:
Hi! You can redownload file and update it.

closed 2020-04-16 by makerbase-mks