makerbase-mks/SGEN_L #21: Firmware

christoels, 2020-03-11:
Hi i cant find the firmware for this setup anywhere have a mks 1.0 controller board in my wanhao duplicator 5s mini printer. With the 3.2 tft screen. I dont know much about firmware and have been trying for a week now to get printer to work. Can anyone please please assist would like to use my printer

makerbase-mks, 2020-03-11:
You use MKS SGEN_L + MKS TFT32 ? Can you send some image about board and tft32 connect and so on

christoels, 2020-03-11:
Hi yes here is the stuff done on the machine the information i have:

MOTHERBOARD - MKS Gen-L 1.0 Controller Board

TOUCH SCREEN - MKS TFT 3.2 inch

POWER SUPPLY - 400W, 12V, 33A

WIFI - MKS ,TFT, Wifi Module

HEATED BED - MKS Mosfet Switch for Heated Bed, 25A

BUILD VOLUME – 205 x 305 x 185 mm

MAX PRINT SPEED – 300 mm/s

BOWDEN EXTRUDER - The new MK10

REMOVABLE GLASS – Bulldog clips

ENCLOSED PRINTER – Perspex doors

LAYER CAPABILITY – 0.02 - 0.3 mm

FILLAMENT CAPABILITIES – PLA, ABS, PETG and other common filaments

CONNECTIVITY – USB stick, USB cable, WIFI, MKS External SD Card Reader

NOZZEL – micro swizz 4.0mm

makerbase-mks, 2020-03-12:
Hi! The first, you need upload firmware to MKS Gen_L V1.0 and set it: 
[image]
The second, connect MKS TFT32 by AUX-1 interface and check temp by MKS TFT32 screen. If it can be detected normally, the communication with the motherboard is successful.

christoels, 2020-03-12:
See i dont know how to do all this dont know anything about coding

christoels, 2020-03-12:
Also dont know which firmware

makerbase-mks, 2020-03-16:
There is firmware in the SGEN_L folder, you can download it and update it.

closed 2020-09-17 by makerbase-mks