makerbase-mks/SGEN_L #24: sensor less homing tcm2209

helgjohn, 2020-04-09:
hi have some issues sensor less homing not working. have followed the video tutorial but no luck getting it running right. Software used is Marlin 2.0.5. 3 Machine Tevo Tarantula PRO Board MKS S Genl 32 bit Drivers TMC2209 uart (mks v2.0) Have followed the steps in tutorial video referred from github site Makerbase but no luck so fare to get it to work. here is what i can due whit sensor less homing enabled i can move all axis and heat bed and so on and no fult or error is reported in display or pronterface every thing seem to be ok. but when i try to home separat axis like x and y only z axis move a bit and it stands idle home all axis then it moves and assume that the possession that print head is current is the home pos. so that means that it moves way out where it shod not. What am i missing??

borland1, 2020-04-09:
On the Marlin Github, there was similar issue reported. Issue # 17204.  Was resolved with rewiring.

helgjohn, 2020-04-09:
> On the Marlin Github, there was similar issue reported. Issue # 17204. Was resolved with rewiring.

Forgive me but i am not sure what you mean here cant find this issue sorry for being a novice at this ....

borland1, 2020-04-10:
Sorry, at the MarlinFirmware Issues tab, you can enter the issue # I gave in the search box to find report.  Here's the link.. 

[https://github.com/MarlinFirmware/Marlin/issues/17204](url)

helgjohn, 2020-04-10:
Thanks for the tips will test it was hoping to avoid wiring but i guess for a budget board......

mks-viva, 2020-04-16:
Hi! MKS TMC2209 V2.0 sensor less homing
1. SGEN_L board Driver IC Power select 3.3V
2. Only set X and Y sensor less homing, Z try not to use
3. The configuration.h driver type is TMC2209
4. Open configuration_adv.h: Enable sensor less homing
5. Configuration_adv.h set X_STALL_SENSITIVITY and Y_STALL_SENSITIVITY value set it 110-130

helgjohn, 2020-04-16:
Ok thank`s i found that the version of my drivers are not mks v2 like i orded but v1 and ver 1.0 they need to be wired in order to work . So then there are the problem to my issue.... Always shit when you can not trust sellers that deliver what they announce for sale and gladly send the wrong parts. 
And this is no mistake from seller its just misleading marketing and a shitty altitude from sellers that is common in many budget online sales platforms. 
Sorry if i wasted som ones time whit my query that shod have been solved by me in house if i had done my home work,,, at least i learned some thing here and maybe others use this to solve issues. Thanks again for good answers

closed 2020-04-16 by helgjohn

borland1, 2020-04-16:
I agree, MKS marketing makes no sense.  Drivers should be V1.0.1 instead of V2.  Only difference I see is V1 is missing 2-pin header.
After you solder on the missing 2-pin header and cover up silkscreen label by installing the heat sink, you will be only one that knows the difference.