makerbase-mks/SGEN_L #25: Can' install Marlin and can't connect to computer

TheMath38, 2020-04-10:
Hi ! I recenty bought one of those's MKs SGEn L with TMC2209.
I configured marlin and compiled it, which was successful. Then, I uploaded it ion the SD card renammed REARM like the Marlin tutorial page. And then... Nothing.

My MKS TFT28 Screen is on, but I  can't move any axis, get the temperature etc etc. Ho, and when I try to connect my main board in my computer,, the USB port is not recognized by Windows!

Please, With this bug I can't use my printer at all ! :(

mks-viva, 2020-04-16:
Hi! About USB driver 
[image]

TheMath38, 2020-04-16:
Hi!
I already tried to install them, but when I try to do it, I have an error message which tell me that I can't because of a missing numerical signature...
Also I don't think that it's a driver problem, because I can't use it by my MKS TFT 28 V4.0 (I've already set my SERIAL_PORT to 0 and SERIAL_PORT_2 to -1 in Marlin...)

Ho, and I forgot to mention that when the mainboard boot up, I have I've the LED D6 which shine in RED constantly, with the LED D1 shining on BLUE too.

Hop this can help

makerbase-mks, 2020-04-16:
[image]
If you connect MKS TFT, you set SERIAL_PORT_2 to 0

TheMath38, 2020-04-16:
Ouupss, sorry, I made a mistake x__x

I already have this option activated, I just switch the settings while writing...
[image]

But yeah, It does not fiw the problem, I mean... I can't boot up on Marlin...

I tried with a different firmware, SmoothieWare. In this case, I can see temp level on the MKS TFT, but I can only move axis in one side only (pretty weird)

NeoMod, 2020-07-16:
I can confirm that the board works as expected (Win10 and Visual Studio) once the parameters are correct. 

If you are still having problems, maybe there is some error in the upload and/or in the procedure. 

Otherwise, if the problem is solved, would you kindly mark the issue as "solved"? And, if possible, let us know what the problem was: it may help other users. Thank you.

Godshenchman, 2020-12-17:
I have the same problem - SGEN_L doesnt show on win7 or linux mint. But the SD card in the SGEN does show. We need "truthful" and proper instructions here. marlin src HAL LPC1768 win usbdriver is nothing - not  ".exe"  Further more as stated it dose not work on linux mint. Yet my robin nano picks up on win7. MSK advertise "CH340" - I think something on the board needs to be flashed - please do something legitimate that works - I note the board has pins right of SD saying GND-TRST // GND TCK // 3.3v TMS  also twin pin top right of chip has "boot."  Gives us instructions of what to do exactly as the board does not have the proper driver. Im certain I can get in running stand-alone but want to use pronterface. Im putting notice on facebook as I have suspicions.

NeoMod, 2020-12-17:
@Godshenchman I had the same problem with a Version 1 of the SGEN_L and unfortunately, as per your intuition, this particular situation occurs when the bootloader of the motherboard gets corrupted during flashing/updating of the firmware.
I'm not sure about all the possibilities, but in my case it was due to an improper setting of the advanced memory features for queuing and buffer.
Anyway, as you have guessed the only viable path is to restore the bootloader but it's not an easy task.
Here you can grasp the general idea, although this example relates to the much simpler 8bit board: https://makerhacks.com/arduino-bootloader-unbricking/
Unfortunately in this case beeing the LPC1768 a 32bit chip with a different architecture the things get more complicated.
A relatively easy overview of the process can be found here, although it's not an MMS board: https://m.youtube.com/watch?v=HQuMuaWjzpM
But I suggest reading also the LPC1768 official documentation which contains some examples of ISP. 
Fair warning: I did end up switching to a new board, since the V2 had just been announced and had some features I was missing. 
The SGEN L v1 with the damaged bootloader is still keeping dust in a drawer, waiting for better days when I could dedicate more time to ISP.

DanJunior78, 2020-12-17:
[image]

I can also confirm,that it's working with TMC 2209 on Marlin 2.0.7. There is a problem to connect sometimes. So, be aware to have access to the card for future updates

I'm also exchanging the board to a BTT SKR Pro. This board is terrific. Sorry for this comment,if MKS is reading it.

Godshenchman, 2020-12-17:
Thanks for your reply. Will look into it. But obviously it looks like they made SGENL V1 without care or regard knowing SGEN L V2 had next to everything & V1 looks like a money spinner for them to gather dust in our draws. I will reward them by going bigtreetech instead.