makerbase-mks/SGEN_L #26: TMC2209 Bi-Directional UART Master/Slave Address Configuration, Marlin 2.0

borland1, 2020-04-12:
When using MKS TMC2209 driver modules, it is not clear how to set up MKS_SGEN_L board jumpers for bi-directional addressing with MS1 and MS2 ( MS1_ADDR0 and MS2_ADDR1 ) jumpers.  

Marlin 2.0 provides ability to setup addressing ( 0 thru 3) for up to four TMC2209 stepper drivers in Configuration_adv.h file, in section 'tmc_smart'...   Address Table:

` * Address | MS1  | MS2
  *       0 | LOW  | LOW
  *       1 | HIGH | LOW
  *       2 | LOW  | HIGH
  *       3 | HIGH | HIGH`

by modifying these code lines as appropriate:

#define  X_SLAVE_ADDRESS 0
#define  Y_SLAVE_ADDRESS 0
#define  Z_SLAVE_ADDRESS 0
#define E0_SLAVE_ADDRESS 0

The MKS_SGEN_L schematic shows:
- MS1 pulled low to ground through resistor Rm1. 
- Rm1 is not identified on PCB silkscreen. 
- Diagram also shows EN pin pulled high by resistor Rm1 to VDD.

I have installed only two MKS TMC2209 modules for axis X and Y with this address scheme:

#define X_SLAVE_ADDRESS 0
#define Y_SLAVE_ADDRESS 2

Where is resistor Rm on the PCB?  And is there any way to modify the board to enable adding additional addressable MKS TMC2209 modules?

closed 2020-04-12 by borland1

NeoMod, 2020-08-28:
@borland1 did you managed to solve the issue?

borland1, 2020-08-28:
> @borland1 did you managed to solve the issue?

Yes.  What I missed is on the schematic, the motor driver slots are shown as typical (only one shown) with label "UART", however there is actually UART1 (E0), UART2 (X), UART3 (Y), UART4 (Z), and UART5 (E1).
So, each TMC driver has it's own separate UART line and so all installed TMC drivers can use same addresses set by driver pins MS1 and MS2.  No address conflicts will be present.
I only installed two TMC2209 drivers for X and Y.