makerbase-mks/SGEN_L #27: Can't install Marlin either

vperso, 2020-04-22:
Hello

I recently bought a MKS SGEN L. I connected it into my 3D printer and tried to update it with a Marlin 2.x firmware, either one already compiled or one compiled  with configuration equivalent to those described in the datasheet (from 2.0.0 to 2.0.5.3).

The firmware seems to be updated since the firmware.bin is transformed into FIRMWARE.CUR.
But what is the LED meaning during this firmware update ? Is there a meaning ?
I refered to the smoothieware troubleshooting guide for meaning but it should be meaningful after the firmware starts.

In my case, I see D6 being red. I see D1 being ON, D2 OFF then D1 OFF and D2 OFF and nothing happens then.
After reset or switch off/on, all led are OFF. Meaning a firmware crash ?
So is the firmware even being booted ?

Thanks for any information around this.

V

makerbase-mks, 2020-04-24:
Hi!
This is normal, because the LED will flash when the motherboard is updating the firmware
1. If you using Marlin firmware. When the Marlin firmware is updated, you will see that only one LED flashes, because marlin only defines one LED for observation.
2. if you using Smoothieware firmware, you will see multiple LEDs blink
These are due to firmware settings and have nothing to do with the hardware.

Kyosuke33, 2020-05-22:
I have a similar problem, i cannot make my board to work.
I use Marlin 2.0, when i switch on, D6 red led and D1 bleu are ON, but nothinh happen.
After reset, D6 red is on and all the blue are on.
When i connect to PC, i see Smoothie.
Look like the board won't install Marlin, but my firmware.bin change to firmware.cur, so normally it's good :/

makerbase-mks, 2020-05-26:
Hi! @Kyosuke33 You need to update your firmware on the SGEN_L motherboard.

Kyosuke33, 2020-05-26:
Hello, thank for the answer. I found the problem. The firmware was good, but a bad drivers on the E avoid the board to start. I remove it and it work fine. Thank you.

makerbase-mks, 2020-05-29:
Ok! That looks like everything is fine.

closed 2020-05-29 by makerbase-mks