makerbase-mks/SGEN_L #28: Upload firmware through USB

flygja, 2020-04-28:
Hi, is there a way to upload firmware through USB? I see some guides on other 32-bit boards saying that the controller's SD card slot will appear as SD card device on Windows 10 - copy and paste firmware then reset the printer. But doesn't seem to work for SGEN L? I see "Marlin SDCard 01  USB Device" in device manager but its not enumerated on windows explorer. I see the disk in diskmgmt.msc but it shows no media even when I have a microSD card plugged in.

Thanks.

makerbase-mks, 2020-05-26:
Hi! You can copy the firmware to the TF card on the motherboard through the computer. The computer can directly access the TF card on the motherboard.

closed 2020-06-03 by makerbase-mks

gukaixu, 2020-07-14:
Hi Flygja I'm facing the same problem with the TMC 2130 drivers on the SGEN_L, is there any way you could share your configuration file with me? Much appreciated (sorry this seems to be the only way I can contact you)

flygja, 2020-07-15:
Hi gukaixu, I haven't been able to get it to work. Even with microSD card plugged into the motherboard, there is no folder that appears in windows explorer for the printer.

gukaixu, 2020-07-15:
Thanks for taking the time to reply flygja. I initially want to test without SPI mode, so just in "normal mode" with the 3 jumpers configured. Do you think it is still needed to define the TMC_USE_SW_SPI? I did try this but unfortunately no success for me...   

 /**
   * Software option for SPI driven drivers (TMC2130, TMC2160, TMC2660, TMC5130 and TMC5160).
   * The default SW SPI pins are defined the respective pins files,
   * but you can override or define them here.
   */
  #define TMC_USE_SW_SPI
  #define TMC_SW_MOSI       -1
  #define TMC_SW_MISO       -1
  #define TMC_SW_SCK        -1`

flygja, 2020-07-17:
Oh you were asking about the previous issue I posted. I'm not sure if need to define TMC_USE_SW_SPI if not using SPI mode, I didn't try it in standalone mode. Sorry.