makerbase-mks/SGEN_L #29: unable to flash firmware via sd, no usb connection

huso4life, 2020-04-28:
I cant get the board to work. I was trying to flash the marlin firmware from https://github.com/makerbase-mks/SGEN_L/tree/master/Firmware/Marlin-bugfix-2.0.x via SD card. I only config changes i made to the config.h file are the follwing:
•filament 1.75
•reprap discount full graphic lcd
•dual z steppers

Marlin was build with vscode+platformIO and stored in a 8gb SD card. On power up the board leds are indicating that something is going on. After a full power down and power up, all leds are on, none are flashing. The firmware.bin on the sd card got replaced by firmware.cur, but:
•all board led's are on
•only the lcd backlight is working but nothing gets displayed
•no usb connection with win7, win10 or octoprint is possible. The board nor the sd card gets recognized over usb

I also tried different configs without success. Also the smoothieware firmware files are not working. Any suggestions on how to get the board to work? Help is greatly appreciated.

paulc1218, 2020-05-03:
i am having the same problem with mac.  my computer doesnt seems to recognise the printer

mentaldemise, 2020-05-12:
Make sure you have the correct serial port(s) defined. I re-used a config from RAMPS that had it as 0 instead of this. If you're using the TFT then you also need to turn on the other one:
#define SERIAL_PORT -1

mentaldemise, 2020-05-12:
Also make sure you don't have the 2.0 instead of the 2. I thought I had the same as the other but the other I had was a 2.0 and some of the pins are different.

DrewBatchelor, 2020-05-21:
Try another SD card. 
I had a SD card that was showing up fine when I plugged it into the S Gen_L
I could access  as a drive over the USB on my pc, and read the files. But it was not being read by the board itself, so config and firmware were not updating.  
Drew

Kyosuke33, 2020-05-22:
I have the same problem with my MKS Sgen L and i don't find the problem :'(

huso4life, 2020-05-26:
I've tried different sd-cards with a capacity from 8 to 32gb. The 32gb card worked well in a pi 3b+. Also I've tried all my usb cables. So far no success. I want to finish my printer so i am returning the board today.

closed 2020-05-26 by huso4life

Kyosuke33, 2020-05-26:
I had the same problem and for me the solution was a bad driver on the E. I replace it and it work great. Try ti remove all driver and test. If doesn't work, try remove motor/endstop/sonde. Maybe one product is faulty like my driver.