makerbase-mks/SGEN_L #30: please help  , the board does not show up on Arduino/Cura.

paulc1218, 2020-05-03:
Hi printers ;) 

i have just received a printer which uses the MKS gen L , when i was trying to upgrade the firmware to marlin, my computer doesnt seems to recognise the printer , nothing is showing up on the Arduino ide nor on any slicers i use. 

i have tried : 
1. using different cables, 
2. download 340 driver (not sure if i am using the right one)
3. try to connect to a different printer (which works) 
4. disconnecting the TFT 

am i missing something here ? any advise is appreciated.  

thanks in advance

borland1, 2020-05-04:
This depository is for MKS_SGEN_L boards. There is a separate depository for that board elsewhere on github.

If you meant you have a MKS_SGEN_L board, then you might try manually updating your windows driver from your  Marlin archive downloaded files.  The driver file is in the HAL folder inside the folder LPC1768.

makerbase-mks, 2020-05-26:
Hi! 
1. CH340 driver https://raw.githubusercontent.com/makerbase-mks/Software/master/CH340G_USB%20Driver/CH340G_USB%E9%A9%B1%E5%8A%A8.rar
2. You need check you port and Baud rate is the same

closed 2020-08-19 by makerbase-mks