makerbase-mks/SGEN_L #31: What are the Diag jumpers doing ?

Flo082002, 2020-05-14:
What is the function of the X_Diag, Y_Diag, Z_Diag, E0_Diag and E1_Diag jumpers?
They are not connected in the circuit diagram.  What are they for?
Regards Flo S

borland1, 2020-05-15:
They connect the TMC driver DIAG output pin which supplies diagnostic and StallGuard information.  High output upon stall motor detection or driver error.
Some later TMC stepper drivers have DIAG pins pre-installed for controller boards like MKS_SGEN_L that have DIAG pin headers on each driver slot.  The row of empty DIAG jumpers are there to disable the feature.  If you are using TMC drivers (like TMC2209) for sensorless homing, then you need to install jumpers, one for each associated axis.   This allows using the drivers without the need to modify/remove the driver's supplied DIAG header pin.

DrewBatchelor, 2020-05-21:
Good Question, Sorry, I don't know the answer either.  I would like to expand this issue. The documentation explaining ALL of the jumpers is missing from Datasheet and documentation for the SGen_L V2. Can it please be added. Thank you. Drew

makerbase-mks, 2020-05-26:
Hi! @Flo082002 @DrewBatchelor 
The DIAG signal is suitable for the TMC2209 / TMC2130 SENSORLESS_HOMING function. Why do we keep jumpers?
When you no longer use the SENSORLESS_HOMING function, the signal from DIAG will interfere with the signal of the mechanical switch, resulting in no return to zero operation or being affected, so you need to remove the jumpers on DIAG. When you use the SENSORLESS_HOMING function, you only need to use the jumper short circuit

closed 2020-06-03 by makerbase-mks