makerbase-mks/SGEN_L #32: PWM Pins for RGB Led Strip

Bubolz, 2020-05-23:
I need to implement LED lighting using an analog RGB LED module,  but only has 2 pins with free PWM for servos ... what other PWM-capable pins could I use?

Thanks!

makerbase-mks, 2020-05-26:
Hi! I want to know, do you use EXP1?

Bubolz, 2020-05-26:
Yes I will use LCD MKS Min, but I will only use one hotend. Could I use the two servo pins (P1.23 and P2.0) and the HE1 pin (P2.6)?

makerbase-mks, 2020-05-29:
of course! Yes you can use it.

closed 2020-06-03 by makerbase-mks

Bubolz, 2020-09-01:
Is there any more PWM pins on the limit switches that can be used? maybe in X +?

makerbase-mks, 2020-09-02:
No, because limit switch has pullup resistor, but you can use LED1-LED4 pin
[image]
[image]

Bubolz, 2020-09-02:
Right! Thanks for the quick response.

I need a pwm that already has a pin for connection. On some machines I’ll use touch LCDs so I’ll have EXP1 and EXP2 free,   previously you suggested using EXP,  what are the available pins?

[image]

makerbase-mks, 2020-09-03:
[image]
You can use P3.25 or P3.26