makerbase-mks/SGEN_L #35: Adding RGB control

Power3DPrinting, 2020-08-05:
I have some "neopixel" 24V RGB 5050 light strips one of 12 LED's and another of 6 LED's wired to 24V power and individually controlled by servo pins P1.24 and P1.25. The LED's turn on with the printer because of the 24V power but I am not getting any start up light control except when powering off the printer they fade in different colors. Any ideas what to do?

/**
 * RGB LED / LED Strip Control
 *
 * Enable support for an RGB LED connected to 5V digital pins, or
 * an RGB Strip connected to MOSFETs controlled by digital pins.
 *
 * Adds the M150 command to set the LED (or LED strip) color.
 * If pins are PWM capable (e.g., 4, 5, 6, 11) then a range of
 * luminance values can be set from 0 to 255.
 * For Neopixel LED an overall brightness parameter is also available.
 *
 * *** CAUTION ***
 *  LED Strips require a MOSFET Chip between PWM lines and LEDs,
 *  as the Arduino cannot handle the current the LEDs will require.
 *  Failure to follow this precaution can destroy your Arduino!
 *  NOTE: A separate 5V power supply is required! The Neopixel LED needs
 *  more current than the Arduino 5V linear regulator can produce.
 * *** CAUTION ***
 *
 * LED Type. Enable only one of the following two options.
 *
 */
//#define RGB_LED
//#define RGBW_LED //Dan: Power ARM uses individual addressible LED's with full range of color

#if EITHER(RGB_LED, RGBW_LED)
  //#define RGB_LED_R_PIN 34
  //#define RGB_LED_G_PIN 43
  //#define RGB_LED_B_PIN 35
  //#define RGB_LED_W_PIN -1
#endif

// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED //Dan: I guess the Power ARM uses neopixels because I don't have enough free pins for individual colors lol
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB //NEO_GRBW // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     P1_24 //P1_26 //4       // LED driving pin
  #define NEOPIXEL2_TYPE  NEO_GRB //NEO_GRBW //NEOPIXEL_TYPE
  #define NEOPIXEL2_PIN    P1_25 //5
  #define NEOPIXEL_PIXELS 12 //30       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 255 //127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup //Dan this is for lighting debug

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

mks-viva, 2020-10-15:
@Power3DPrinting I has test RGB function and i use the latest marlin bugfix 2.0.x, it is work, but i use MKS SGEN_L V2 motherboard
- LED will change to red when current temp = target temp
- LED will change to white when printing
- LED will change to green when print finished

mks-viva, 2020-10-16:
https://github.com/MarlinFirmware/Marlin/pull/19762
But MKS SGEN_L V1 have no pins with P1_19/P1_20/P1_21

mks-viva, 2020-10-20:
@Power3DPrinting I will close this question temporarily.

closed 2020-10-20 by mks-viva