makerbase-mks/SGEN_L #37: MKS SGen L V1 with LCD12864B slow screen refresh

DanJunior78, 2020-08-20:
Hello gents,

I've a problem with the refresh of the LCD screen. It's behaving like a laptop screen below 5 degrees Celsius or above 50 degrees Celsius.
I'm not aware of a possible setting in Marlin Firmware to increase the refresh rate. On my other board, it's a Gen L with the counterpart display version A...all is running smooth.
Does anyone have an idea?

NeoMod, 2020-08-28:
@DanJunior78 sorry for bothering you, but I'm having troubles getting the same LCD on the same Board work with Marlin 2.0. May I ask what version of Marlin are you running on the MKS SGEN L ?

As per your issue, if it's still unsolved have you tried enabling `#define LIGHTWEIGHT_UI` ?

DanJunior78, 2020-08-28:
@NeoMod no problem. It's running on Marlin 2.0.5.4 . Usually the LCD is easy to activate. You've only to activate the MKS mini display. I had at the beginning the same problem as I had 2 display drivers activated. But I've not tried to use the lightweight UI as it should be running with the standard UI, like my Gen L which is running the same firmware (I've only swapped the motherboard in the settings).
If MKS would answer,it might help, because I'm running on the SGen the TMC 2130. On the Gen only the TMC2208.

DanJunior78, 2020-08-28:
If you want to give it a try...there is a bin file and the Marlin source...for display testing it might be enough...as i mentioned. I´m using the TMC2130 with SPI.

https://www.dropbox.com/s/g8ld2f8rhiy8z99/MKS-SGen-master.zip?dl=0

makerbase-mks, 2020-09-17:
https://github.com/makerbase-mks/SGEN_L/issues/38

DanJunior78, 2020-09-17:
Could @makerbase-mks do a test to verify that the problem is caused by the SPI connection to the TMC2130?
I've tried to compile the Marlin 2.0.6.1 and using the drivers in normal mode there is no overload on the SPI and the display works fine.
I'm little bit unhappy not being able to use the SPI functionalities for the drivers.

DanJunior78, 2020-09-17:
Yes, I had also this phenomenon, when I tried to use the TMC2130. Depending on the Marlin version.

makerbase-mks, 2020-09-18:
I think, the SPI driver of LPC176x needs further improvement and it's just a matter of time.

closed 2020-10-14 by mks-viva

DanJunior78, 2020-10-14:
Is it working now? There is a conflict in the SPI, because the SPI for the LCD and e.g. TMC 2130 run on the same bus/connection. Will there be a workaround?