makerbase-mks/SGEN_L #38: Marlin 2.0 LCD12864B - Blank Screen, only backlight

NeoMod, 2020-08-28:
### Bug Description

After correctly compiling/flashing Marlin 2.0 with LCD support, the LCD won't display anything. The backlight is on (meaning it receives the correct voltage) but nothing shows up.

The buzzer emits a continuous high-pitched tone, and the "Stop" button seems to be working when pressed.

### My Configurations

Motherboard: MKS SGEN L v1
LCD: MKS LCD12864 Model B
Marlin: 2.0.6.1

### Steps to Reproduce

1. Followed the instructions available on the User Manual of the SGEN L to compile the firmware with LCD Support. Tried also the configuration suggested on the product description page.
2. Flashed the new firmware; printer works as expected, TFT also works.
3. LCD shows a "blank screen"; no bootlogo is shown while powering up the printer. 

**Expected behaviour:** [What you expect to happen]

I expected the LCD to show the actual Marlin Menu, or at least an image of it until I adjusted the contrast. 

**Actual behaviour:** [What actually happens]

LCD backlight turns on, the buzzer is active emitting a constant low-frequency pitch, and nothing is shown on the screen.

#### Additional Information

* Tried with both #define MKS_LCD12864 and #define MKS_MINI_12864
* Tried both with and without TFT connected.

NeoMod, 2020-08-28:
Edit: I just noticed this reply from MKS in the Marlin Repository BUG Tracker. 

https://github.com/MarlinFirmware/Marlin/issues/19159#issuecomment-682461269

It seems there is a problem with the latest Marlin release.

makerbase-mks, 2020-09-02:
Yes! This is a bug, but the cause has not been found yet. You can use Marlin 2.0.1 first, and it's no problem

makerbase-mks, 2020-09-17:
@NeoMod The latest marlin bugfix-2.0.x has fix it, you can download it and update.
https://github.com/MarlinFirmware/Marlin/pull/19269
https://github.com/MarlinFirmware/Marlin/pull/19271

closed 2020-09-17 by makerbase-mks