makerbase-mks/SGEN_L #40: tried to update drivers and marlin

m6cac, 2020-09-24:
Have tried to update drivers to TMC2208 had a lot of trouble installing firmware but now have managed.
I've enclosed some files so you can see what I have done.
I
[Marlin-2.0.6.1.zip](https://github.com/makerbase-mks/SGEN_L/files/5279759/Marlin-2.0.6.1.zip)

 now have just a blank screen and nothing else

DanJunior78, 2020-09-25:
I could show you my version using TMC 2209 in UART mode and it's working fine.

m6cac, 2020-09-25:
I used youtube MKS SGEN L V1,0 - TMC2208 UART install for my guide

DanJunior78, 2020-09-25:
If it's an old video, there might be some differences, e.g. in Marlin 2.6.1 there is the LCD12XXXB available. You shouldn't use the MKS mini.
Do it stepwise. First the board. Then display. Then the drivers first in standalone then try in UART mode. You'll find the step you do something wrong.

m6cac, 2020-09-25:
I have MSK MINI 12864 V2.1
Do you know the setting I need for this?

DanJunior78, 2020-09-25:
Search in the configuration.h for the MKS_Mini_12864 and enable it. There should be only one LCD active. Otherwise it's not working.

m6cac, 2020-09-26:
I've had some luck. I've had some movement using external software Wrong movements but a start. still no joy with the display.
I'll try something different tomorrow.

makerbase-mks, 2020-09-26:
@m6cac 
Use MKS MINI12864: #define MKS_MINI_12864 in configuration.h file, but you need download the latest version marlin bugfix-2.0.x, because it has fix MKS MINI12864 and MKS LCD12864B display

m6cac, 2020-09-26:
Have now managed to get a display. on the display at the bottom, it says TMC connection error.
this learning is stressful.

DanJunior78, 2020-09-26:
Lol, keep trying. Which TMC 2208 you have? Did you jumper correctly the board? To use the UART mode you need to change the jumpers.

m6cac, 2020-09-26:
Have 1 jump for UART on each driver.

DanJunior78, 2020-09-26:
Check with M122 (you have to release the TMC debug in config_advanced) if all are not recognized or only some/one is broken. I had this topic with the TMC2130.

DanJunior78, 2020-09-26:
And some TMC 2208 have to be soldered to be able to run in UART mode. Which TMCs brand have you bought? Watterot, MKS,...?

m6cac, 2020-09-26:
Thank you for your help. I'm going to try something else. I'll leave this post open for another day and hopefully, I can give some results.

m6cac, 2020-09-26:
Managed to get results in standalone, But now have a problem with heaters.

closed 2020-09-27 by m6cac