makerbase-mks/SGEN_L #42: SGEN_l marlin filament outrun sensor issue

hardy8, 2020-10-03:
Hello, I'm trying to connect filament runout sensor on MKS SGEN-L board with TFT-35 touch screen. I made necessary changes to use filament sensor on board connection but my TFT-35 screen doesn't seem to like that very much.
While filament sensor connected on board, when i pull out the filament print is stopped and nozzle parked the determined position as it should be, but TFT-35 doesnt react this move and acts like print still continues. Sad thing is while filament is pulled out and nozzle is parked, i tried to stop/pause on TFT screen but nothing happens. Seems like runout sensor and TFT not communication while sensor connected on board.
TFT has its own pins allocated for filament sensor but i need to customise printhead moves for my printer, so TFT pins are not an option for me.
Do i necessarly use TFT pins or can i find a way to make TFT talk to Marlin 
(firmware TFT35 v1.0.7, Marlin-2.0.7)

Thanks,

mks-viva, 2020-10-14:
Hi! You need connect filament runout sensor to TFT35， because your printing file by TFT35’ SD card.

mks-viva, 2020-10-26:
Is this problem solved?

closed 2020-10-27 by mks-viva