makerbase-mks/SGEN_L #43: Extruder Fan doesn't turn on automatically above 50 F

alcore23, 2020-10-14:
Good afternoon,

I had to reinstall my firmware on my MKS SGEN L, which I have installed on and Ender 3, and after the update the extruder fan won't come ON automatically when the hotend is above 50 F. 

I see that the Temperature menu now shows Fan 1 and Fan 2 as well, and I can manually turn the fans from this menu option. Fan 1 turns the part cooling fan ON, and Fan 2 turns the extruder fan ON.

I have not changed the wiring and this function worked fine on my older firmware, any recommendations?

mks-viva, 2020-10-14:
Hi! Can you provide your config to me?

alcore23, 2020-10-14:
Yes

[Marlin.zip](https://github.com/makerbase-mks/SGEN_L/files/5379461/Marlin.zip)

alcore23, 2020-10-14:
I am using the latest on this repository, which I believe is 2.0.7

Demitryk, 2020-10-14:
> Which Marlin Firmware are you running. On mine with 2.0.6 it's working
> fine. Hotend Fan connected to the E1 heating output.

you need to set  "#define E0_AUTO_FAN_PIN" to "FAN1_PIN"

DanJunior78, 2020-10-14:
It's like dimetry wrote. Something in your config wrong.

alcore23, 2020-10-14:
> 
> you need to set "#define E0_AUTO_FAN_PIN" to "FAN1_PIN"
> […](#)

This did the trick! Thank you very much all.

I was confused because the notes said that -1 would disable it all together so I had tried commenting out that line, but I didnot try to assign a value.

closed 2020-10-14 by alcore23