makerbase-mks/SGEN_L #44: SGEN_l V1.0 & MKS MINI12864 compatiblity

bacco70, 2020-10-26:
hello, i've both this and try to use but seem to not compatible 
marlin 2.0.7.2
in configuration.h 
#define MKS_MINI_12864 as active but if i connect to the board they have a full backligth and beeper always on

Demitryk, 2020-10-26:
You have to rotate the exp connectors

If the screen is not turning on, flip the exp1 connector.

If the knob is not working and it seems to be malfunctioning, flip the exp2
connector

I have detailed instructions on thingyverse, look in the mks SGen guide.
You may not have a Tevo but the instructions for the screen are the same

https://www.thingiverse.com/thing:4626471

bacco70, 2020-10-27:
board sgen_l v1.0
dispaly mini_12864 v2.0
i think we can add in pins_MKS_SGEN_L.h something like this:
#if  ENABLED(MKS_MINI_12864)
  #define BEEPER_PIN                     P1_31
  #define BTN_EN2                        P0_18
  #define BTN_EN1                        P0_15
  #define SD_DETECT                      P3_27
  #define SD_SCK                         P0_07
  #define SD_CS2                         P0_28
  #define SD_MOSI                        P0_09
  
  #define BEEPER                         P1_31
  #define DOGLCD_A0                      P1_00
  #define BTN_ENC                        P1_30
  #define DOGLCD_CS                      P0_17
  #endif
but where i must place ,to do this authoritative?

bacco70, 2020-10-27:
with these setting work , but in compiling output there are many many redefinition error:
 #if  ENABLED(MKS_MINI_12864)
  #define MISO_PIN                       P0_08
  #define BTN_EN2                        P3_25
  #define BTN_EN1                        P3_26
  #define SD_DETECT_PIN                  P0_27
  #define SD_SCK                         P0_07
  #define SS_PIN                         P0_28
  #define SD_MOSI                        P0_09
  
  #define BEEPER_PIN                     P1_31
  #define DOGLCD_A0                      P1_00
  #define BTN_ENC                        P1_30
  #define DOGLCD_CS                      P0_17
  #endif

and in the menu miss "SDCARD PRINT"

mks-viva, 2020-11-04:
Can you provide me with the entire marlin firmware?

bacco70, 2020-11-04:
how?
its 2.07.2, with only config e config_adv edited

mks-viva, 2020-11-04:
OK! I will test and it is working and i provide it for you.

mks-viva, 2020-11-04:
Use MKS MINI12864 on MKS SGEN_L V1 motherboard
1. Inconfiguration.h file
  setting #define MOTHERBOARD BOARD_MKS_SGEN_L
  Enable  #define MKS_MINI_12864

You can download firmware.zip and update it to your board
You can download Marlin-bugfix-2.0.x.zip source code(has setting use MKS MINI12864) complie it
[firmware.zip](https://github.com/makerbase-mks/SGEN_L/files/5485812/firmware.zip)
[Marlin-bugfix-2.0.x.zip](https://github.com/makerbase-mks/SGEN_L/files/5485813/Marlin-bugfix-2.0.x.zip)

bacco70, 2020-11-04:
ok, but i can try on december, i'm away for work
you suggest me to use the bugfix version instead the 2.0.7.2?
the mine lcd it's version 2.0

mks-viva, 2020-11-11:
@bacco70 
It is will wroking?

bacco70, 2020-11-11:
@mks-viva
hello, until 31 nov i can't try, i'm away from house

mks-viva, 2020-11-13:
OK, wait for you to test it, and it can work normally.

bacco70, 2020-12-06:
OK, now work correct, thanks
during boot there are many pixel casually on,before appear Marlin logo: there are some way to keep pixel off before Logo?

mks-viva, 2020-12-07:
Haha, there is currently no way to deal with it, it looks more like LCD’s own hardware startup.

mks-viva, 2020-12-11:
I will close this issues temporarily

closed 2020-12-11 by mks-viva