makerbase-mks/SGEN_L #45: TMC2208 UART problem

lucasgomezortiz, 2020-11-11:
How about, I have a sgen l board and some tmc2208 from the manufacturer geeetech. I can't put the uart mode with the tmc, it marks me "connection error" I followed the steps described here but I can't make them work, in standalone mode they work without problem.
If I have doubts, the github guide does not say anything about putting any jumper on the X-uart connectors

Here I leave what the pronterface gives me when I enter M122

		X	Y	Z	E
Enabled		false	false	false	false
Set current	600	600	800	800
RMS current	1049	1049	1436	1436
MAX current	1479	1479	2025	2025
Run current	18/31	18/31	25/31	25/31
Hold current	9/31	9/31	12/31	12/31
CS actual	0/31	0/31	0/31	0/31
PWM scale
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false
msteps		256	256	256	256
interp		false	false	false	false
tstep		0	0	0	0
PWM thresh.	790	790	5270	4347
[mm/s]		100	100	3	30
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	0	0	0	0
pwm scale auto	0	0	0	0
pwm offset auto	0	0	0	0
pwm grad auto	0	0	0	0
off time	0	0	0	0
blank time	16	16	16	16
hysteresis
 -end		-3	-3	-3	-3
 -start		1	1	1	1
Stallguard thrs
uStep count	0	0	0	0
DRVSTATUS	X	Y	Z	E
sg_result
stst		*	*	*	*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

lucasgomezortiz, 2020-11-11:
Something I forgot to clear these up using the Marlin version bugfix-2.0.x

mks-viva, 2020-11-11:
Hi! Can you provide your tmc2208 picture and your buy it link to me, i will try to and how to use it.

lucasgomezortiz, 2020-11-11:
> Hi! Can you provide your tmc2208 picture and your buy it link to me, i will try to and how to use it.
This is the tmc, also something that I notice with this manufacturer, that the colors do not correspond to the position, if I look for DIR in the MKS it is inverted with respect to the guide colors that the TMCs have

Blue (tmc) --- Black (board)
Black (tmc) --- Green (board)
[image]

mks-viva, 2020-11-11:
Ok! You need setting it
1. MKS SGEN_L motherboard setting UART mode and use jumpers
[image]
2. Connect PDN-UART to pin4
[image]

mks-viva, 2020-11-13:
@lucasgomezortiz 
Hi! It is will working?

lucasgomezortiz, 2020-11-15:
> @lucasgomezortiz
> Hi! It is will working?
sorry for the delay in answering, just these days I started to check the printer, I already made the short but it still does not work, "TMC CONNECTION ERROR"

closed 2020-11-18 by lucasgomezortiz