makerbase-mks/SGEN_L #46: Problem with Mks Sgen_L V1 and TFT35

acca84, 2020-11-12:
Hello, 
I have a problem for printing on Mks Sgen_L V1 with TFT35 from USB (or SD)
The printer is a cartesian printer with two hotends

I select my gcode, start print and sometimes bed and printheads preheat and temperatures freeze on the display a little bit before the desired temperature, print never starts. At this point, i can still navigate in the LCD menus but it has no effect on the printer.
I have to reset printer after that (OFF/ON)

Sometimes, the first layer is printed then the printer stops (temperatures freeze on the LCD like in the first example).  

I didn't print many parts but i managed to print parts using only one hotend, never parts using two hotends.
The printer always stops at the same moment for a same gcode.

Here is an example of what i test print : two rectangle parts (4mmx4mmx2mm)
[image]
when using two hotends, the beginning of the gcode is the following (only the beginning because for this code, printer heats but never starts printing, hotend1 is stuck at 192/200°C, always the same) : 

`
;FLAVOR:Marlin
;TIME:225
;Filament used: 0.141155m, 0.0234001m
;Layer height: 0.1
;MINX:33.4
;MINY:59.4
;MINZ:0.3
;MAXX:65.6
;MAXY:138.8
;MAXZ:2
;Generated with Cura_SteamEngine 4.7.1
T0
M140 S60
M105
M190 S60
M104 S200
M104 T1 S175
M105
M109 S200
M105
M109 T1 S175
M82 ;absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E10
G92 E0
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:8
;LAYER:0
M107
G0 F2400 X38.87 Y61.383 Z0.3
;TYPE:SKIRT
G1 F1500 E0
G1 F1200 X39.561 Y60.999 E0.03944
G1 X40.286 Y60.687 E0.07882
G1 X41.039 Y60.45 E0.1182
G1 X41.813 Y60.291 E0.15762
G1 X42.598 Y60.21 E0.19699
`

Then exactly the same printing but using only hotend 1 (no problem with this one)  : 

`;FLAVOR:Marlin
;TIME:114
;Filament used: 0.0770429m, 0m
;Layer height: 0.1
;MINX:35.2
;MINY:60.2
;MINZ:0.3
;MAXX:64.8
;MAXY:79.8
;MAXZ:1
;Generated with Cura_SteamEngine 4.7.1
T0
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E10
G92 E0
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:8
;LAYER:0
M107
G0 F2400 X38.87 Y61.383 Z0.3
;TYPE:SKIRT
G1 F1500 E0
G1 F1200 X39.561 Y60.999 E0.03944
G1 X40.286 Y60.687 E0.07882
G1 X41.039 Y60.45 E0.1182
G1 X41.813 Y60.291 E0.15762`

Thanks for your help and sorry for my bad english

mks-viva, 2020-11-13:
Hi! Can you provide the voides when printng to me?

acca84, 2020-11-13:
Hi
Do you mean you want the gcodes ?
Thanks

acca84, 2020-11-13:
[CFFFP_cube 4x4x1 One hot end.txt](https://github.com/makerbase-mks/SGEN_L/files/5535216/CFFFP_cube.4x4x1.One.hot.end.txt)
[CFFFP_cube 4x4x1 Two hotends.txt](https://github.com/makerbase-mks/SGEN_L/files/5535217/CFFFP_cube.4x4x1.Two.hotends.txt)

acca84, 2020-11-14:
Today, i tried to print the same Gcode with pronterface over USB, it did print the whole code
I think the problem comes from the screen

lucasgomezortiz, 2020-11-15:
> Today, i tried to print the same Gcode with pronterface over USB, it did print the whole code
> I think the problem comes from the screen

in that case try using another SD card

acca84, 2020-11-15:
I tried with the TFT with SD and usb

acca84, 2020-11-20:
Hello, i found the origin of the problem
I plugged a LCD 12864 screen to the mks sgen l board and found that the print stopped due to a thermal error.
It was caused by a wrong PID values for hotend 1
I modified PID and now the printer prints correctly.
Would it be possible to show this kind of errors on the TFT 35 ?
Thanks

DanJunior78, 2020-11-20:
I've same problem,but I've both connected, LCD and the TFT. The TFT is really useless.
The new generation TS35 is much better,combined with the Robin board and using Marlin UI, the new colorful one.

mks-viva, 2020-11-23:
@DanJunior78 
Haha! I also agree with the point of view, because the MKS Robin Nano V2+MKS TS35 method supports more UI interfaces, simpler operation, and more cost-effective.

DanJunior78, 2020-11-23:
@mks-viva, to be honest, it's like it should be. I'm printing only via server and a screen should be like...first look and you see your printer state. This is little bit the problem on the MKS UI and mainly the integration of the TFT35. You have to click to see the temperature. You don't see,for manual tests, the X and Y position, for sure first you have to go to movement and you see the Z Axis.so, I'm stick to the Marlin UI on the TS35. I can't say too much on SD card prints.
If you compare to the BTT TFTson the other hand, MKS is far behind. They have a really nice integration of their UI. All visible on the first, main, screen, temperatures, position and fan. By the way...the switching between the modes is on the fly and not related to the firmware. 
(Should be only a small pinch to the MKS R&D, ;o) ).

mks-viva, 2020-11-23:
@DanJunior78 
Thank you very much for your feedback. In your opinion, in the MKS UI style, what other functions need to be improved or added?

DanJunior78, 2020-11-23:
@mks-viva , i think the firmware parameters are all in, as i believe to remember, i´ve seen most of them. The visual should be improved to have on the main screen the main values.
Here you see the TFT35...it stays like this. Always, if you print via server
[image]

This is usually, similar to the LCD view, what most which i know, would like to see (SD card prints i would agree to have the special view, if choosen)
[image]
[image]
[image]
Now the gadget on BTT screens.
[image]