makerbase-mks/SGEN_L #47: Problem with MKS_SGEN_L_V1 as replacement of MSK GEN L on Artillery Genius

palopjay, 2021-01-12:
I set up hardware and setup MARLIN according to last MARLIN 2.0 version, assigned right TMC2100. It compiles succesfully, all systems works except all axis. If Homing X, Y or Z, the only axis is moving is the 2nd Z axis (E1 on Main Board) loudly and fastly.

mks-viva, 2021-01-13:
Hi! Can you provide a vodies whne homing?

palopjay, 2021-01-13:
Thank you, I attached the video. For all axis homing, the result doesn't change: same movement, same noise, same error. The extruder moves separately, noisy also that but is the only axis is "indipendent". 
Paolo

https://user-images.githubusercontent.com/77364400/104411002-d34e8e80-5569-11eb-9709-7ccd58b5de5f.mp4

mks-viva, 2021-01-13:
I know what caused it
1. You need to set the true or flase of the limit switch
2. Z axis looks like DIR is wrong

palopjay, 2021-01-13:
I tried M119, and all limit switch are open. I tried inverting them and I got them triggered so I think they are ok as opened. And:I pressed X homing in the video, and only the right Z motor moved. I expect the X motor moves.. Thanks!

palopjay, 2021-01-13:
I tryed to tune up all switches and axis directions, inverting them and re-invert because of no results. I saw Z1 motor turns faster than Z2 motor. The TMC have same jumpers for NORMAL MODE, so may be the marlin files for MKS are damaged on GITHUB? 
Does someone still use this MAIN BOARD to share the last running configuration? I tested the vref and looks good on all axis (X-Y=0.620V  /  Z1-Z2=0.680V) , so the difference between Z axiz i deriving from motherboard settings.
Thank you

DanJunior78, 2021-01-13:
Are you sure you configured the z drives right in Marlin?

palopjay, 2021-01-14:
Hi. I'm almost sure. I have the same configuration I had in the old MKS_GEN_L was on printer. But the big doubt is that also if I send the HOME for X axis by printer face, the axis moves is Z. The same for X axis. Every homing input moves z axis. I tried also to use, as base of Marlin, the suggested upgrade repository for my printer (differs for TMC, but I changed them in Marlin Configuration, https://artillery.n3t.ro/upgrades.html) but for all Marlin version used I got same Only Z axis movement.

mks-viva, 2021-01-14:
Yes, for the new version of marlin firmware, when going home, Z-lift is performed first, and then XY home is performed.

palopjay, 2021-01-14:
I saw that Z2 stepper is the 1st Extruder free.
```
// The Z2 axis, if any, should be the next open extruder port
#if NUM_Z_STEPPER_DRIVERS >= 2
  #ifndef Z2_STEP_PIN
    #define Z2_STEP_PIN   _EPIN(Z2_E_INDEX, STEP)
    #define Z2_DIR_PIN    _EPIN(Z2_E_INDEX, DIR)
    #define Z2_ENABLE_PIN _EPIN(Z2_E_INDEX, ENABLE)
    #if Z2_E_INDEX >= MAX_EXTRUDERS || !PIN_EXISTS(Z2_STEP)
      #error "No E stepper plug left for Z2!"
    #endif
```
 E1 in this case. So, if I have:

`#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many`

in my configuraion_adv.h, and I have

```
#define X_DRIVER_TYPE  TMC2100
#define Y_DRIVER_TYPE  TMC2100
#define Z_DRIVER_TYPE  TMC2100
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2100
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2100
```
in configuration.h
Are the only 2 position where to enable and define z2 axis. I loose something?

Thank for your support

palopjay, 2021-01-14:
In addition:
I tested all stepper motors and all Stepper drivers. All of them are ok.

DanJunior78, 2021-01-15:
First you need to do something like an I/O check. Check all switches. Check moving directions of the main drives. I could send you my setting files of Marlin from my SGen-L board in my Anycubic i3 Mega S (running Marlin 2.0.6). The only problems I had and these are problems from the board design, is using the drivers in SPI mode with a LCD 12864 in parallel. Drivers in UART mode work fine. So, my setup is using TMC 2209 with this board. It's running like a charme.

palopjay, 2021-01-15:
I checked all stepper drivers and all stepper motors. They works. Only the exit from motherboard is not giving results. They works in normal mode... All wrongly

DanJunior78, 2021-01-15:
Wrongly means, wrong direction? In normal mode of the drivers all work or not? Are you using SPI mode, correct? So, first, check in normal mode to have all directions correct. Second, if you setup your driver's to SPI mode, activate in Marlin M122 to check what Marlin gets as connection and if all drives are communicating to your mainboard. If not...you found one issue. I had some bad TMC 2130 ...they are like...every 6th driver is broken,when you get them. I would not recommend the TMC 21 series. Get from AliExpress TMC 2009. They are stronger and running smooth.

mks-viva, 2021-01-16:
TMC2100‘s current supported is very small

palopjay, 2021-01-16:
I'm sure TMC2100 are ok for the configuration because they were working good with old MainBoard. 
BUT.. 
I checked that all TCM are facing the same VREF, but have different "EFFECT" on motors (run faster or slower with no reason. i tried 1 by 1 on the same axis this time). So the problem here are the TMC are damaged. I'll change all axis TMC to 2208UART, and I'll solve the problem for sure. Thanks for your help

closed 2021-01-16 by palopjay