makerbase-mks/SGEN_L #49: MKS SGEN_L V1.0

youngdaddy09, 2021-02-20:
Hi MKS SGEN_L board is not connecting to my computer to flash firmware does anybody have the drivers for this for my windows 10 in device manager it shows up in usb an a error not a port so i cannot do anything with it if any one can help me that would be great new to this stuff

sudo-lelek, 2021-02-20:
MKS SGEN uses LPC Chip wich Could Not be programmed via serial. You Need to put Firmware.bin on SD-Card and Stick it in the Board. Then Power on or reset. After leds glow constantly the fw flash is successfull

youngdaddy09, 2021-02-20:
Ok thanks i am all over the place its asking for a upload port an i cannot figure this out  (Looking for upload disk...
*** [upload] USB\firmware.bin: No such file or directory)

youngdaddy09, 2021-02-20:
Ok i figured that part out now all the lights are on an staying on

mks-viva, 2021-02-21:
You need copy firmware.bin file to TF card and insert to board and repower on or reset

tuzlusu62, 2021-06-10:
lütfen bir kişi yardım etsin...Ben aliexspresten "MKS Gen V1.4"  Sürücülerim 8825... Bir yıldır uğraşıyorum...Motorlara hareket veremedim...Ekran kartım MKS TFT 32_L V4.0.....Bu siteyi yeni buldum acemilik işte....Ama ana kartım ile ilgili kısım bulamadım kartımın modeli geride mi kaldı?