makerbase-mks/SGEN_L #50: SGEN_L + TMC2209 stuck in spreadcycle mode

Power3DPrinting, 2021-03-16:
Hi, I am having trouble disabling hybrid and spreadcycle mode and quieting the printer down. Right now it is as loud if not louder than my A4988+RAMPS 3D printer and nowhere near other printers with TMC2130's or TMC2209's. I have 5xTMC2209's enabled in configuration.h. In confuguration_adv.h hybrid mode is commented and the hybrid threshold is at 900 mm/s for good measure. Micro stepping is set at 16 and 16 step interpolation is true. M122 TMC Debug reports all drivers and motors are ok. Even the z motors moving at 4 mm/s sound like A4988 drivers.

DanJunior78, 2021-03-17:
If they are recognized, they should work. I am using the same combination on mine and it's working. Fystec TMCs needed some cutted pins on the TMC.
You can play around on the LCD with the values.

Demitryk, 2021-03-17:
Which brand and version of 2209s do you have?

Power3DPrinting, 2021-03-17:
> Which brand and version of 2209s do you have?

Thanks, those are good checks,

I am using TMC2209 V2.0 MKS bought together with some SGen_L V1.0 all from makerbase. I have not cut any pins on the drivers or modified them in any way other than attaching their heatsinks. I have jumped the UART pin below (and nothing underneath) all 5 drivers and jumped the diag pins on all but E0 DIAG.

I appreciate seeing the guide, I checked that they matched the SGen L manual on github. I did learn some suggestions like I should increase stall sensitivity from 60 to 150 or bump distance should be {0,0,0} instead of {5,5,5}.

DanJunior78, 2021-03-17:
I'm using the SGenL v2 (I believe) in my Any i3 S. The TMC 2209 are Fystec V3. I've connected a LCD 12864.
The TFT is really useless and a waste of money. I've it connected parallely to the LCD.
With the TMC you can have following problem.
Some have soldered the Resistor that they are running in standalone mode. This is not your Problem, because you get feedback by the M122.
So, if you're running in UART mode...I believe you won't be able to change it by this messy TFT

Demitryk, 2021-03-17:
now are they loud while printing or only during homing? When using sensor-less homing, they home in spread-cycle, you cant change that. its requirement for sensor-less homing. 

most people only use sensor-less homing on X and Y. having them on the Z is a bad idea, did you remove all you endstops?

Power3DPrinting, 2021-03-17:
Pardon my parrot in the background, this is the noise with reduced acceleration:
https://www.youtube.com/watch?v=mseaGdQ7tpI

This is the noise with default acceleration, it's really not even moving that fast which is confusing:
https://www.youtube.com/watch?v=TE6FGtqvSy8

I want to look forward to the V2, it looks nice especially with the extra fan mount is handy. As for the TFT, I've seen plenty of printers work with them..? It's my first touch screen on a printer. They are weird and enjoying it. Thanks, that's a relief, it's good to know I don't have to use a soldering iron.

That makes sense about the spread-cycle when homing! What is odd that I am noticing is that the X is very quiet when homing, for whatever reasons the Z and Y are abnormally loud. I removed the Z and E1 diag pins and commented Z_Stall_sensitivity but that has introduced a homing error where M119 says all end stops are open but the printer immediately detects a collision so I am still working on that to get it working to test it out.

1kemanciii, 2021-05-14:
> I'm using the SGenL v2 (I believe) in my Any i3 S. The TMC 2209 are Fystec V3. I've connected a LCD 12864.
> The TFT is really useless and a waste of money. I've it connected parallely to the LCD. [...]

How do we set up for tmc2209 v3 ?

DanJunior78, 2021-05-15:
Depends how you're going to use them. Standalone or UART or with sensorless homing. There is a video from MKS

1kemanciii, 2021-05-15:
uart mod. but TMC Connection Error

DanJunior78, 2021-05-15:
If you've placed well the jumpers it should work. If not try in standalone mode. If they are working there, it might be some soldering necessary on the TMC.

Power3DPrinting, 2021-05-16:
> If you've placed well the jumpers it should work. If not try in standalone mode. If they are working there, it might be some soldering necessary on the TMC.